GlenNicholls/solar_tracking_project #1: 'top' and regex causes py.test to give false failure during unit testing

GlenNicholls, 2018-10-13:
Using top command in the system_monitor.py class causes py.test to hang:

`

# get current cpu usage in %
    def get_cpu_use_perc(self):
        cpu_use = os.popen("top -n1 | awk '/Cpu\(s\):/ {print $2}'").readline().strip('\\')
        print('-DBG- CPU Use call from terminal returns: {}'.format(cpu_use))
        self._cpu_use = float(cpu_use)
        return self._cpu_use

` 

This is the output of the test failure. looks like it might be annoying to get py.test to support top:

`

                                                                                                                       
_____________________________________________________ test_memory _____________________________________________________
                                                                                                                       
                                                                                                                       
num_checks = 20                                                                                                        
                                                                                                                       
    def test_memory(num_checks=20):                                                                                    
        print('-I- ----Memory Information----')                                                                        
        string = []                                                                                                    
        string.append('CPU [%]')                                                                                       
        string.append('RAM [%]')                                                                                       
        string.append('DISK [%]')                                                                                      
        max_str_len = len( str(max(string, key=len)) )                                                                 
        num_cols    = len(string)                                                                                      
                                                                                                                       
        write_table(string=string, max_str_len=max_str_len, header=True)                                               
                                                                                                                       
        for i in range(num_checks):                                                                                    
>           cpu_use_perc, ram_use_perc, disk_use_perc = get_memory()                                                   
                                                                                                                       
src/system_monitor/test/test_sys_mon.py:128:                                                                           
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
                                                                                                                       
src/system_monitor/test/test_sys_mon.py:53: in get_memory                                                              
    cpu_use_perc  = sys_mon.get_cpu_use_perc()                                                                         
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
                                                                                                                       
                                                                                                                       
self = <system_monitor.system_monitor.system_monitor object at 0xb58a6ad0>                                             
                                                                                                                       
    def get_cpu_use_perc(self):                                                                                        
        cpu_use = os.popen("top -n1 | awk '/Cpu\(s\):/ {print $2}'").readline().strip('\\')                            
        print('-DBG- CPU Use call from terminal returns: {}'.format(cpu_use))                                          
>       self._cpu_use = float(cpu_use)                                                                                 
E       ValueError: could not convert string to float:                                                                 
                                                                                                                       
src/system_monitor/system_monitor.py:50: ValueError                                                                    
------------------------------------------------ Captured stdout call -------------------------------------------------
                                                                                                                       
-I- ----Memory Information----                                                                                         
| CPU [%]  | RAM [%]  | DISK [%] |                                                                                     
-DBG- CPU Use call from terminal returns:                                                                              
------------------------------------------------ Captured stderr call -------------------------------------------------

`

bandtank, 2018-10-14:
I can't tell what's going on because the formatting of your post seems to be alternating between code and non-code tags*. Could you please fix the formatting? Check the preview window to see if it looks right. You may need to use 4 spaces instead of backticks to trigger the code format tags.

GlenNicholls, 2018-10-14:
Whoops, sorry about that, I thought my last update fixed it. It is updated correctly now!

The `get_cpu_use_perc` function is called, it returns an empty string in the py.test run. After checking the other sys_mon.py tests, they all work, so it is not an issue with `os.popen()` as all of those tests pass. Looking at the documentation, it looks like there are special rules so I think getting CPU usage needs to change, or I'll have to look into using a marker or something to get the test to work with the command.

bandtank, 2018-10-14:
One thing that should probably be changed is subcommands return value parsing. For example, here's the prototypical output of vcgencmd:

    (01:08 PM) tony@Thor:~ $ sudo /opt/vc/bin/vcgencmd measure_temp
    temp=48.9'C

However, what happens if permission is denied, the command doesn't run, or something else goes wrong? The current parser is trying to remove specific invalid characters instead of extracting what's known to be valid; in other words, use a whitelist instead of a blacklist. Here's a regex that will be more specific without being sensitive to line endings (portability) and other issues:

    /temp=([\d\.]+)'C/i

If the regex matches, capture group 1 will have a valid floating point value. Otherwise, it should be None, which you can use to make decisions about how to respond. I suspect part of the issue with `get_cpu_use_perc` is related to parsing unexpected string values among other things. Here's what I see when I run a simpler version of your command:

    >>> os.popen("top -n1").readline()
    '\x1b[?1h\x1b=\x1b[?25l\x1b[H\x1b[2J\x1b(B\x1b[mtop - 13:23:35 up 11 days, 17:10,  3 users,  load average: 0.00, 0.03, 0.10\x1b(B\x1b[m\x1b[39;49m\x1b(B\x1b[m\x1b[39;49m\x1b[K\n'

Using a regex like this:

    /load average\s*:\s*([\d\.]+)[\s,]+([\d\.]+)[\s,]+([\d\.]+)/

to get three capture groups that give you this:

    $1: 0.00
    $2: 0.03
    $3: 0.10

For example:

    >>> import os
    >>> import re
    >>>
    >>> str = os.popen("top -n1").readline()
    >>> m = re.search('load average\s*:\s*([\d\.]+)[\s,]+([\d\.]+)[\s,]+([\d\.]+)', str)
    >>> m.group(0)
    'load average: 0.08, 0.02, 0.06'
    >>> m.group(1)
    '0.08'
    >>> m.group(2)
    '0.02'
    >>> m.group(3)
    '0.06'

As an aside, you may benefit from a `get_all` function in the system_monitor class, e.g.:

    def print_all_params(self):
      print("Temp C = {:.2f}".format(self._temp_C))
      print("Temp F = {:.2f}".format(self._temp_F))
      print("Temp K = {:.2f}".format(self._temp_K))
      # etc.

GlenNicholls, 2018-10-29:
used psutil for this. Still need to kick off test to ensure this works correctly. Will also fix the tab issue from the stupid editor at school being trash:

https://github.com/GlenNicholls/solar_tracking_project/blob/0b5ccb8a39f6578742e7fd70900c9cc86290d15f/src/system_monitor/system_monitor.py#L47-L53

GlenNicholls, 2018-11-01:
Per the following code, this is resolved by using psutil. I will probably re-factor to use this instead of all of the command line stuff in this package.

https://github.com/GlenNicholls/solar_tracking_project/blob/33bd25d69f85e850662e37234e8bfcc10e950139/src/system_monitor/system_monitor.py#L56-L64

closed 2018-11-01 by GlenNicholls