GlenNicholls/solar_tracking_project #4: Multiple interrupts causing contention

bandtank, 2018-10-17:
https://github.com/GlenNicholls/solar_tracking_project/blob/89bd5b87050819dc284a78d5802128b977265ed5/misc_dev/attiny84a/src/power_circuit_ctrl/power_circuit_ctrl/main.c#L239

How will multiple interrupts cause contention? This comment makes me think the right way to handle servicing the power control lines is to do it in a function outside of an interrupt. Setting flags and making a decision later will prevent contention issues if you don't have all the information in one ISR.

GlenNicholls, 2018-10-17:
That was an old comment that I don't think applies. I think the most complicated scenario is that the RTC alarm state interrupt is currently expected to change while the pi hold on interrupt is being serviced. I think this should change as it doesn't feel like the right way of doing it. As for contention, I've just been reading the value of the PORTA register to make sure it's being changed only when it needs to be.

bandtank, 2018-10-17:
We can talk through the logic to decide how to handle the actual setting and clearing of the bit. There are a number of ways to do it, most of which involve either an explicit or implicit mutex. Implicit mutexing can be accomplished by abstraction of error and/or boundary checking in functions.