GlenNicholls/solar_tracking_project #6: Software debounce

bandtank, 2018-10-17:
https://github.com/GlenNicholls/solar_tracking_project/blob/89bd5b87050819dc284a78d5802128b977265ed5/misc_dev/attiny84a/src/power_circuit_ctrl/power_circuit_ctrl/main.c#L314

Implementing a software debounce using a timer will be relatively simple and usually quite robust. I'm guessing you didn't literally mean sleep, but, just to be sure, don't use sleep. The ISR will sit there twiddling it's thumbs which is bad news. ISRs need to get in and get out ASAP.

GlenNicholls, 2018-10-18:
So when using a timer, I assume it should just be started in the ISR, then the processor goes back to sleep and wakes up again when the timer raises a software interrupt?

GlenNicholls, 2018-10-18:
Oh nevermind, I see there is a timer interrupt vector.

bandtank, 2018-10-18:
You can start and stop timers wherever you want. Timers may be one of, if not the most, useful peripherals. It would behoove you to study the timer chapters in depth because the counting patterns, waveforms, match generation interrupts, and tons of other things have practically unlimited applications. I once made a state machine that used four timers, each with four compare match interrupts, to drive an external DAC and then the input capture source of a fifth timer to get nanosecond precision on a timing pulse. I've seen much crazier things from other designers as well.

Tiny's don't have a ton of features, but the timers are still powerful. Once you get a handle on things, come back to timers to keep the system asleep as long as possible in as many situations as possible. You can double up on compare match interrupts for multiple high-level functions by keeping track of rollovers and things like that in globals or static variables.