GlenNicholls/solar_tracking_project #7: Readability and maintainability enhancements

bandtank, 2018-10-17:
https://github.com/GlenNicholls/solar_tracking_project/blob/89bd5b87050819dc284a78d5802128b977265ed5/misc_dev/attiny84a/src/power_circuit_ctrl/power_circuit_ctrl/main.c#L292

In general, doing things like that can be prone to bugs later. It's also quite difficult to change pin assignments if you decide to make layout modifications. Do something like this:

    #define POWER_ON_PIN  PA0
    #define POWER_ON_PORT PORTA
    #define TURN_POWER_ON  SET_BIT(POWER_ON_PORT, POWER_ON_PIN)
    #define TURN_POWER_OFF CLR_BIT(POWER_ON_PORT, POWER_ON_PIN)

That is much, _much_ easier to maintain and it's also a lot more clear while reading the code what is happening. It also protects against logic inversions, e.g. if the power pin is active low instead of active high. Turning it on is very clear whereas literally setting or clearing a bit may not be.

GlenNicholls, 2018-10-18:
Is it best practice to place all of these in a header file?

bandtank, 2018-10-18:
That’s a bit of a religious question depending on who you ask. In general, yes, this type of thing should be in a header file because the code shouldn’t have any idea about the physical manifestation of the algorithm. Turn the power on means turn the power on regardless of how that translates into hardware. It’s not always that concise obviously, but abstraction is your friend especially when it comes to hardware interfaces.

Typically, I create header files with the following things from top to bottom:
* include statements
* hardware specific preprocessor definitions
* macros like LED0_ON, etc.
* function prototypes

In more complex projects, I will often put hardware specific definitions in a separate file to keep things organized and obvious. It depends on a lot of things, so there’s no right answer; be consistent and clear regardless.

While I’m thinking about it, embedded applications have to be smart about using strings more than you may currently realize. When you printf a fixed string, for example, it stores the string on the stack or heap depending on the situation, but you can’t afford to to support either option.

The flash memory is in a different address space depending on the architecture and accessing it from an instruction may or may not be possible. If it is possible like in an Atmel microcontroller, you can store strings there at the cost of significantly increased access time. On small microcontrollers, avoid this entirely by not using strings at all. When I say strings, I mean `char *` just to be clear.

In terms of memory conservation, globals and general purpose registers can be quite useful as long as you understand the volatile keyword and all that it implies. People get their parties in a bunch about globals on any platform, but they’re fine even in high level languages as long as you use them properly. Anyway, embedded applications are different due to memory constraints, so the paradigm isn’t as taboo.