GlenNicholls/solar_tracking_project #18: Consider using built-in macros for AVR register access

bandtank, 2018-10-21:
https://www.nongnu.org/avr-libc/user-manual/group__avr__sfr.html

```
#define _BV(bit)   (1 << (bit))
#define bit_is_set(sfr, bit)   (_SFR_BYTE(sfr) & _BV(bit))
#define bit_is_clear(sfr, bit)   (!(_SFR_BYTE(sfr) & _BV(bit)))
#define loop_until_bit_is_set(sfr, bit)   do { } while (bit_is_clear(sfr, bit))
#define loop_until_bit_is_clear(sfr, bit)   do { } while (bit_is_set(sfr, bit))
```

It could simplify some of the defines, but it's fine either way.

GlenNicholls, 2018-10-21:
interesting, I'll check this out! I knew there was something like this, but the examples I saw online looked like they were for Arduino IDE, so I wasn't sure where those were coming from.

bandtank, 2018-10-21:
On that note, check the [library reference](https://www.nongnu.org/avr-libc/user-manual/modules.html). There's a ton of stuff you might need, e.g. [Power Reduction Management](https://www.nongnu.org/avr-libc/user-manual/group__avr__power.html). There's value in learning how to do these things manually, but using library functions doesn't completely mask the requirement to understand what you're doing with the hardware.

GlenNicholls, 2018-10-22:
If I have to learn how to use a library, I quit.

bandtank, 2018-10-22:
Let’s go collect sand so we can make our own silicon wafers. I think we should start all the way at the beginning to make sure we do as much non-innovative work as possible.

GlenNicholls, 2018-10-22:
Ooo, that's a good idea. Another thing I thought would be a good idea for system configuration, we should use a CSV with sub-delimeters to load our settings

GlenNicholls, 2018-10-25:
Used many of these noted in the macro libraries.

closed 2018-10-25 by GlenNicholls