GlenNicholls/solar_tracking_project #19: Define clocking for uC in Makefile

GlenNicholls, 2018-10-24:
https://github.com/GlenNicholls/solar_tracking_project/blob/5e9910b60f4fe3ec53cd6ad5541e268fe9e95c2b/misc_dev/attiny84a/src/pi_hat_ctrl/Makefile#L23-L24

Best to define clocking here and pass it down to the code... Spent a long time trying to talk to this device again after reducing the clock to 125kHz and the default programming speed couldn't identify the device or recover fuses.

-i is the spi clock delay, so this should also be parameterized according to system clock. Should be ~4 times slower than the uC system clock

bandtank, 2018-10-24:
You might want to go slower than 4x. That should be fine, but slower won't hurt anything.

Is the delay in microseconds? You should probably put a comment specifying the unit and how it is used to derive the clock frequency. It will also be very helpful down the road if you put a comment in the makefile about how to program the microcontroller using the file. It's fresh in your mind now, but it won't be in six months (hypothetically).

GlenNicholls, 2018-10-24:
Yes, it is in us. I'll put a comment in there to denote both of those things. What is a reasonable frequency to run at? I am currently using 125kHz, but I feel like I could divide the clock all the way and be fine, I'm just not sure if there is going to be something that will bite me later.

bandtank, 2018-10-24:
Which clock do you intend to use as the system clock and at which frequency? As far as I know, this system doesn’t have much processing to do and there are relatively lax deadlines involved; assuming that’s true, I think the low power, internal 128 KHz oscillator would be sufficient. You may even be able to use a prescaler of 8 to provide a system clock of 16 KHz if the system can tolerate that setting.

In the first case, I would use a 25 KHz programming frequency. In the second case, 4 KHz would be appropriate. 25% is a good rule of thumb and should always work, but there’s no reason to rush it since this will be done very infrequently, if ever, after it works reliably.

Debug is different. For now, go faster so you aren’t waiting forever to iterate.

GlenNicholls, 2018-10-29:
documenting your email suggestion so I don't forget to do this:

    make -DF_CPU=125000

somewhere in the code:

    #ifndef F_CPU
    #error You must define F_CPU
    #else
     #if F_CPU == 125000
       OCRA1 = 0x12;
     #elif F_CPU == 250000
       OCRA1 = 0x34;
     #else
       #error Unsupported value for F_CPU
     #endif
    #endif

GlenNicholls, 2018-10-30:
From suggestion, this has been tested and works well. The worst case programming SCK has been accounted for to be 4x slower than the lowest device frequency:

https://github.com/GlenNicholls/solar_tracking_project/blob/64621bb102d3db1e32168b8d61db00277a282412/misc_dev/attiny84a/src/pi_hat_ctrl/Makefile#L24-L30

The below is used to configure each timer along with clocking:

https://github.com/GlenNicholls/solar_tracking_project/blob/64621bb102d3db1e32168b8d61db00277a282412/misc_dev/attiny84a/src/pi_hat_ctrl/attiny84.h#L86-L155

closed 2018-10-30 by GlenNicholls