GlenNicholls/solar_tracking_project #20: Programming uC will cut power to pi

GlenNicholls, 2018-10-25:
When the pi starts to program eeprom by pulling reset low, all the pins enter their default state causing the load switch to turn off. This will mean the pi will be shutoff as soon as it starts to program. If this is just a floating pin, we can add an external pullup. If not, I think the best solution is to add a jumper for programming mode where the load switch is pulled up and the pi doesn't lose power when programming. Will investigate this further this weekend.

GlenNicholls, 2018-10-25:
If there is a way to make pins use pullups during reset, I will do that instead.

bandtank, 2018-10-25:
No. You will have to use an external component to guarantee the state of a pin that isn't actively driven by the microcontroller. During reset, the pins configured as floating inputs and, as far as I know, there's no way to alter that behavior.

GlenNicholls, 2018-10-25:
Darn, wish I thought of this before we did the PCB. I'll just blue-wire something for the time being and we'll update the schematic to use a jumper.

GlenNicholls, 2018-10-25:
So open the issue against the layout and reference in the notes of the schematic?

GlenNicholls, 2018-10-25:
Also, I just tested this and it looks like the load switch isn't affected. The output remains high for anything other than a logic zero. I'm not sure if it's worth looking into further in this case, but I'm open to suggestions. I guess if the load switch ever gets changed this might be an issue, but as of now, it looks like it is not affected during ICSP.

GlenNicholls, 2018-10-25:
So the datasheet says that it cannot be left floating and doesn't say anything about an internal pullup. I think putting an external pullup on that line makes the most sense unless you would prefer we use a jumper. I'm going to use a pullup for our PCB so we don't have to re-spin it, but we can put a jumper in the schematic/layout we give you.

bandtank, 2018-10-25:
I presumed it would not have an internal pullup, so you'll need to add one. This is a bug waiting to happen.

There's nothing wrong with cutting traces, adding components, or using jumpers to modify the board. It's uncommon to get it right on the first try and correct behavior is far more important than a pristine layout.

GlenNicholls, 2018-10-26:
I see your point. Closed testing at home at room temperature with no switching is fine. I suspect that higher temperatures and noise will cause this to fail.

bandtank, 2018-10-26:
Yeah, that's pretty much the idea. Temperature, noise, stray capacitance, solar activity, and many other things could cause non-deterministic behavior.

GlenNicholls, 2018-10-31:
@graemeDnf, feel free to close this when the pcb reflects ISSUE#20 in the schematic

closed 2018-11-03 by graemeDnf