GlenNicholls/solar_tracking_project #21: How accurate is this?

GlenNicholls, 2018-10-25:
https://github.com/GlenNicholls/solar_tracking_project/blob/5e9910b60f4fe3ec53cd6ad5541e268fe9e95c2b/src/main.py#L35-L43

Need to figure out how accurate this is. If this is pulling location from nearest ISP whatever it's called, this could be pretty far away from customers location.

I messed with something similar (not with python) a long time ago and it was yielding lat/long of Comcast in Colorado Springs about 5 miles from my house. I don't know how this error would affect us if this is the case, but it's worth looking in to. If this is the case and it is too much error, we could have something in the high-level interface for the user to enter their specific coordinates.

bandtank, 2018-10-25:
There's nothing wrong with requiring input from the user to setup a system like this. Getting the location automatically is a nice feature, but I wouldn't bother with it. It's too small of a feature and it will cost too much time to validate the process. Ask for lat/long and call it a day.

GlenNicholls, 2018-11-19:
Just checked this and it is indeed very far off due to info above. A better solution might be http://www.indjango.com/google-api-to-get-lat-long-data/ but this is not a priority at all. I will be hardcoding for now unless we have time to do this