GlenNicholls/solar_tracking_project #23: Switching noise from alarm condition causes uC to briefly enter reset due to TTL threshold

GlenNicholls, 2018-10-29:
RESET_N pin of uC has been having issues when the alarm pin goes low causing device to briefly enter reset which causes glitching on power control pin (led to false positive during testing). Need to add 10K pullup to fix the metastable condition.

Probing without the pullup, the voltage is 1.54V (3.3V Vcc), but drops to 1.48V during an alarm condition before slowly coming back up to 1.54V where it comes out of reset. Probing the RESET_N pin caused this brief reset not to occur, so I think that capacitance caused the device not to drop below the TTL threshold.

bandtank, 2018-10-29:
The reset pin definitely needs to be pulled high. You should also add a
capacitor to ground.

GlenNicholls, 2018-10-29:
We have this going through a level shifter, should the LPF be on the high or low side? Does it matter?

bandtank, 2018-10-29:
Is the schematic updated? If so, where should I be looking?

GlenNicholls, 2018-10-31:
@graemeDnf , feel free to close this when the PCB has the pullup noted by ISSUE#23 in the schematic.

closed 2018-11-01 by graemeDnf