GlenNicholls/solar_tracking_project #25: Sleep mode for attiny84a does not work per noted code

GlenNicholls, 2018-10-30:
After referencing a few sources in addition to the following, the below sleep code still does come out of sleep from interrupts:

[https://www.electronicsweekly.com/blogs/engineer-in-wonderland/atduino-go-sleep-attiny-2017-08/](url)

https://github.com/GlenNicholls/solar_tracking_project/blob/dc396e505067940ffb68f793bffcb7f520a1185a/misc_dev/attiny84a/src/pi_hat_ctrl/main.c#L185-L216

bandtank, 2018-10-30:
Have you tried it without any of the more complex logic? What happens if you simply go to sleep in main using `sleep_cpu()`? Have you ever successfully woken the uc from sleep in any configuration?

GlenNicholls, 2018-10-30:
Sorry for the confusion, I just tested that with just the IDLE sleep mode and the device successfully comes out of that. When I test with power down, I'm unable to get it to come out of sleep with an alarm condition on the RTC which will turn power on based on INT0

bandtank, 2018-10-30:
Make sure you are following the constraints in this paragraph:

```
7.1.3 Power-Down Mode
When the SM[1:0] bits are written to 10, the SLEEP instruction makes the MCU enter Powerdown
mode. In this mode, the Oscillator is stopped, while the external interrupts, and the Watchdog
continue operating (if enabled). Only an External Reset, a Watchdog Reset, a Brown-out
Reset, an external level interrupt on INT0, or a pin change interrupt can wake up the MCU. This
sleep mode halts all generated clocks, allowing operation of asynchronous modules only.
```

GlenNicholls, 2018-10-30:
So I was trying to debug why it wasn't working so I continue to re-enable the interrupts as I wasn't sure if there was something that wasn't stated explicitly in the datasheet (the ads on the site made it impossible to see what that person's reasoning was). After removing that code, I see the same behavior.

Once I added a 50ms delay, I was able to get the push button to get the device out of power down, so after removing the delay, it looked like it would sometimes catch the button interrupt, and sometimes it wouldn't (due to the 8-bit timer clock going down beforehand). However, the `INT0` interrupt was never recognized with this delay. After I used a pin change interrupt, `PCINT10` on `PCINT1_vect`, I am able to capture all interrupts, so it does look like the `cli();` caused the processor to catch the interrupt, but not set it in time before disabling interrupts.

Looks like the issue was with there not being enough delay before `cli();`, but I think that at 32kHz, the `INT0` level is not held long enough with the 50ms delay.

I think this is resolved, but this brings up a question again that I asked you a while ago about putting numerous pin change interrupts on one INT vector. Will this have any implications moving forward that I am not aware of?

GlenNicholls, 2018-10-31:
This has been tested and will now be closed.

closed 2018-10-31 by GlenNicholls