GlenNicholls/solar_tracking_project #33: [PCB] fix missing connection on attiny devmode level shifter

graemeDnf, 2018-10-30:
[image]
Connection was deleted between low side of level shifter and 3.3v rail

graemeDnf, 2018-10-30:
PCB file updated and pushed, awaiting confirmation of bluewire fix on existing board

GlenNicholls, 2018-10-30:
yeah that's gonna be a bit if you want to CALM DOWN

GlenNicholls, 2018-10-30:
Just to confirm, the highlighted via connects to the left pad of the resistor right above it?

graemeDnf, 2018-10-30:
Correct. This is the bottom copper though, so the directions will be flipped when you look at the bottom side of the actual board. Also, it will probably be easiest to solder that resistor pad to the left pad of one of the neighboring resistors (same net) rather than the via