GlenNicholls/solar_tracking_project #69: [PCB] Dev mode and FAULT pins of the attiny never go low due to level shifter

GlenNicholls, 2018-11-16:
due to the resistor on the high-side of the level shifter, both LED 1 & 5 do not turn off as there is 2.75V "pullup" on this net. After probing the low-side, these both always read 3.3V. Is it valid to remove these two resistors??

GlenNicholls, 2018-11-16:
Another weird thing I notice is when I press and hold the button, both LEDs mentioned turn off completely, but the status LEDs for the load switch are unaffected.

bandtank, 2018-11-16:
LEDs on these nets likely won't work long term. I didn't think about it before, but you'll probably need to drive the LEDs with another pin. You could also use comparators to drive the LEDs after calculating the expected voltages on either side of the level shifters, but that's a lot of work for no real gain. You can't pull any of the resistors out or the level shifter will stop working.

GlenNicholls, 2018-11-17:
Hmmm, good to know, I guess it is a good thing this isn't critical, although it will be annoying for the user. Is there a way to make this work with the level shifters?

bandtank, 2018-11-17:
I gave it some thought for a minute or two this morning and I didn’t come up with an obvious solution. Part of the problem is speed. If the net needs to be fast, it would be a bad idea to drive an LED without a buffer in series. There may be a more elegant solution, but using a comparator would work. You’d need to know the voltage levels in every condition: driving high and low from one side and then the other. At that point, pick a side and set the reference.

GlenNicholls, 2018-11-19:
we do have two free pins on the tiny, so I think that is the cleanest solution. @graemeDnf once the PCB is updated, feel free to close this

bandtank, 2018-11-19:
I typically put LEDs on a shift register. Even if you don’t want to use SPI, you can clock a shift register manually to get 8+ LEDs with only two pins. Just an idea.

graemeDnf, 2018-11-29:
Moved relevant LEDs from level shifter lines to free GPIO pins (BCM 0 and BCM15)

closed 2018-11-29 by graemeDnf