GlenNicholls/solar_tracking_project #102: [PISW] Thank god I caught this

GlenNicholls, 2018-12-07:
This calibration function should be revised. It works as is right now, but it should use interrupts like before. Make sure that az/el are isolated so the state of one doesn't influence the other. 

https://github.com/GlenNicholls/solar_tracking_project/blob/d01d46178745f1d25ad69ea53f6527f5529b52ae/src/motor_control/stepper_motor.py#L94-L104

GlenNicholls, 2018-12-07:
NOTE: The interrupts are currently disabled for the debugging on tracing this issue earlier. They need to be re-enabled for the system to recognize when it reaches the limit switch limits. Currently, the software will be unable to detect the az/El limits outside of calibration. I will not be touching the code before tomorrow, so I am logging this here so it doesn't slip through the cracks.

bandtank, 2018-12-07:
When you make temporary changes, put TODO in a nearby comment.

Then run this every so often to make sure you aren’t forgetting anything:

grep -ri todo *

Also, do not commit temporary changes. Use this to commit pieces of files so you can leave the temporary change in the working copy instead of the staged or committed copy:

git add -p [files]

Never commit a temporary change without some form of tracking - todo comments, partial commits, tasks/issues, etc.

GlenNicholls, 2018-12-07:
I see, thanks for the suggestions, I'll keep those in mind moving forward.

On campus, it's hard because it isn't possible to scp wirelessly, so the easiest way to develop there has been to push/pull, but I hate doing that because it just clutters the commits with debug code. Is there a better way to work between systems using the -p flag or something similar?

bandtank, 2018-12-07:
If you absolutely have to commit, use one of the other two methods to track
the temporary code and put it in a branch that is known to have bad code.
Then you can leave it out of a merge into master. I would avoid this at all
costs, though, because once it's in the repo it will eventually be used
incorrectly, unknowingly, or both.

bandtank, 2018-12-07:
Another option is to build debug hooks into the code.

```
if debug:
  #do stuff
else:
  #do other stuff
```

It's still not ideal, but it's much, much safer.